nebrius/raspi-i2c #6: Symbol is not defined

Workshop2, 2015-08-09:
Hello,
I am trying to get johnny-five up and running on my RPI 2, but I get "Symbol is not defined" in index.js from this lib.

I can see raspi-io importing Symbol, but this lib doesn't seem to have a dependency or import for it - is this correct?

See: https://github.com/Workshop2/initio-js

Thanks,
Simon

nebrius, 2015-08-10:
Thanks for writing in! I tried to do a quick fix for you, but it didn't work 😞. I'm going to have to take some time to dig into what's going on here. In the mean time, if you switch to Node.js 0.12, this issue will go away: http://node-arm.herokuapp.com/

So a little more info about Symbol (if you're already familiar with symbols, feel free to ignore the rest of this comment). Symbols are a new feature in ECMAScript 2015, more commonly called ES6. They are built in to the JavaScript language itself, for JS engines that support it. The version of V8 that comes with Node.js 0.12 supports symbols, but the version that ships with Node.js 0.10 does not.

Symbols are a way of getting _actually_ private variables on objects, instead of prepending an underscore to the name and pretending they're private. It's a really nice feature, and one I use rather extensively. The ```import 'es6-symbol/implement';``` statement in raspi-io is a polyfill for this feature. Admittedly, I only test on Node.js 0.10 when an issue such as this one comes up, so these things tend to slip through the cracks. Eventually, I plan to set up a CI environment that will automatically test these things on various versions of Node.

Workshop2, 2015-08-11:
Aha, yes that makes sense. I did a dodgy patch on my copy to import the ES6 Symbol package.

I am trying to get a simple PMW motor working through RPI/Johnny 5 (http://4tronix.co.uk/store/index.php?rt=product/product&keyword=initio&category_id=0&product_id=171) but I am being plagued at every step.

Thanks for such a detailed response - is this something you are going to patch soon?

nebrius, 2015-08-11:
I'm trying to patch this soon, yes. For some reason NPM segfaults when trying to install es6-symbol on my RPi when running NPM 1.x, so not super fun times fixing this :-/

nebrius, 2015-08-13:
This should be fixed with raspi-i2c 1.0.5 (and raspi-io 3.3.5). Let me know if it fixes things for you, and thanks for the bug report!

closed 2015-10-06 by nebrius