nebrius/raspi-i2c #9: 32 byte limitation

sgoll, 2016-08-03:
Currently `raspi-i2c` limits all payloads to 32 bytes which is fine for SMBus (and required by the specs) but not necessary and even limiting for regular I2C.

The problem is the `checkLength` and `checkBuffer` check, respectively, which happens both when running SMBus (supplying `register` parameter to the function) as well as plain I2C (without `register`).

I suggest skipping those checks when `address` is `undefined`. The underlying functions from `i2c-bus` that get called for plain I2C do not require the 32 bytes length, so enforcing this in `raspi-i2c` without being necessary makes it impossible to send or receive long plain (non-SMBus) I2C payloads.

See fivdi/i2c-bus#23 for more information.

sgoll, 2016-08-03:
I created pull request #10 with the suggested changes.

nebrius, 2016-11-25:
Closing since the PR was landed

closed 2016-11-25 by nebrius