nebrius/raspi-i2c #14: Not working with Compute Module 3 :(

Silvia-t, 2019-01-30:
Hi guys,

I didn't read about any limitations when using the raspi-i2c with a compute module, but when trying to run a test program on CM3:

`Unknown board revision a220a0, assuming Raspberry Pi Zerp/2/3 pinout. Unless you are running a compute module or very old RPi this should work fine. Please report this board revision in a GitHub issue at https://github.com/nebrius/raspi-board.`

Is there any way in which the GPIO to be used for i2c could be user-defined? 

Thanks.

nebrius, 2019-01-30:
That warning is expected, as I've never added support for computer modules.

We can't use any GPIO for I2C because of the way I2C works. It doesn't work by setting pins high or low, it works by setting pins low or "high impedance," using what's called an open-drain transistor configuration. As such, you need hardware support to do I2C.

I've never used a compute module before, so I don't know what it exposes. I would assume that the compute module routes the hardware I2C pins to it's external pins though. I'll add though that the Raspi.js suite doesn't support the compute module officially (I should add this to the list of requirements). In the case of I2C though, it should mostly work though.

Do you need to run any other modules from Raspi.js on your compute module? If not, I'd recommend using [i2c-bus](https://www.npmjs.com/package/i2c-bus) directly, which is the module I use to get I2C support.

P.S. I'd like to request that you don't use "guys" to describe us as a group of collaborators. We're not all guys, and culturally as a project we strive to use gender neutral language.

nebrius, 2019-02-01:
@Silvia-t the raspi-i2c API is a very thin shim on top of i2c-bus, and the APIs are really similar. In the I2C Bus documentation, registers are referred to as "commands," which I could see being confusing if you were looking for something titled "register." Hope that helps!

Silvia-t, 2019-02-04:
Thanks for the info @nebrius :)

nebrius, 2019-02-21:
I'm going to close this issue now, but feel free to reopen if you have anymore questions. FYI I did add a note in the project requirements that the compute module is not supported.

closed 2019-02-21 by nebrius