nebrius/raspi-i2c #15: Error: Could not locate the bindings file.

dkossman, 2024-09-04:
I'm getting the following error trying to use raspi-i2c on a Pi 4 running the latest Raspberry Pi OS. This code was working before, not sure what changed, but i had to reinstall everything so probably missed a step. any insights would be helpful. modules installed are:

npm install influx express path body-parser ds18b20-raspi os raspi raspi-i2c pug iconv-lite

my code (poolApp.js) is here: https://github.com/dkossman/poolServer, if you need access let me know and I'll add you as a collaborator.  same problem with a much simpler example.

pi@raspberrypi:~/poolServer $ node -v
v18.19.0
pi@raspberrypi:~/poolServer $ uname -a
Linux raspberrypi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

pi@raspberrypi:~/poolServer $ node poolApp.js
/home/pi/poolServer/node_modules/bindings/bindings.js:135
  throw err;
  ^

Error: Could not locate the bindings file. Tried:
 → /home/pi/poolServer/node_modules/i2c-bus/build/i2c.node
 → /home/pi/poolServer/node_modules/i2c-bus/build/Debug/i2c.node
 → /home/pi/poolServer/node_modules/i2c-bus/build/Release/i2c.node
 → /home/pi/poolServer/node_modules/i2c-bus/out/Debug/i2c.node
 → /home/pi/poolServer/node_modules/i2c-bus/Debug/i2c.node
 → /home/pi/poolServer/node_modules/i2c-bus/out/Release/i2c.node
 → /home/pi/poolServer/node_modules/i2c-bus/Release/i2c.node
 → /home/pi/poolServer/node_modules/i2c-bus/build/default/i2c.node
 → /home/pi/poolServer/node_modules/i2c-bus/compiled/18.19.0/linux/arm64/i2c.node
 → /home/pi/poolServer/node_modules/i2c-bus/addon-build/release/install-root/i2c.node
 → /home/pi/poolServer/node_modules/i2c-bus/addon-build/debug/install-root/i2c.node
 → /home/pi/poolServer/node_modules/i2c-bus/addon-build/default/install-root/i2c.node
 → /home/pi/poolServer/node_modules/i2c-bus/lib/binding/node-v108-linux-arm64/i2c.node
    at bindings (/home/pi/poolServer/node_modules/bindings/bindings.js:126:9)
    at Object.<anonymous> (/home/pi/poolServer/node_modules/i2c-bus/i2c-bus.js:4:32)
    at Module._compile (node:internal/modules/cjs/loader:1356:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1414:10)
    at Module.load (node:internal/modules/cjs/loader:1197:32)
    at Module._load (node:internal/modules/cjs/loader:1013:12)
    at Module.require (node:internal/modules/cjs/loader:1225:19)
    at require (node:internal/modules/helpers:177:18)
    at Object.<anonymous> (/home/pi/poolServer/node_modules/raspi-i2c/dist/index.js:26:19)
    at Module._compile (node:internal/modules/cjs/loader:1356:14) {
  tries: [
    '/home/pi/poolServer/node_modules/i2c-bus/build/i2c.node',
    '/home/pi/poolServer/node_modules/i2c-bus/build/Debug/i2c.node',
    '/home/pi/poolServer/node_modules/i2c-bus/build/Release/i2c.node',
    '/home/pi/poolServer/node_modules/i2c-bus/out/Debug/i2c.node',
    '/home/pi/poolServer/node_modules/i2c-bus/Debug/i2c.node',
    '/home/pi/poolServer/node_modules/i2c-bus/out/Release/i2c.node',
    '/home/pi/poolServer/node_modules/i2c-bus/Release/i2c.node',
    '/home/pi/poolServer/node_modules/i2c-bus/build/default/i2c.node',
    '/home/pi/poolServer/node_modules/i2c-bus/compiled/18.19.0/linux/arm64/i2c.node',
    '/home/pi/poolServer/node_modules/i2c-bus/addon-build/release/install-root/i2c.node',
    '/home/pi/poolServer/node_modules/i2c-bus/addon-build/debug/install-root/i2c.node',
    '/home/pi/poolServer/node_modules/i2c-bus/addon-build/default/install-root/i2c.node',
    '/home/pi/poolServer/node_modules/i2c-bus/lib/binding/node-v108-linux-arm64/i2c.node'
  ]
}

nebrius, 2024-09-04:
Hey @dkossman. Raspi I2C hasn't been updated in 5 years and is no longer being maintained, so I'd imagine that something doesn't work with newer versions of Node.js. You're welcome to fork this repo and see if you can get it building again though.

closed 2024-09-04 by dkossman

dkossman, 2024-09-04:
thanks for the info -sorry for the false alarm - it turned out that the npm install must have been corrupted - I re-ran the rpm installs in a new directory and no longer get this error.   However i am still having problems with the i2c calls - are you aware of a similar npm module for handling i2c from javascript that is being maintained?

nebrius, 2024-09-04:
> thanks for the info -sorry for the false alarm - it turned out that the npm install must have been corrupted - I re-ran the rpm installs in a new directory and no longer get this error. However i am [...]

I'm not aware of any unfortunately. NGL I use C++ for all my iOT stuff these days, not JS.